seong-hun/fym #84: 코드 최적화 (속도 개선)

JinraeKim, 2019-09-04:
현재 50초 시뮬레이션을 돌리는데 수 초가 소요되고 있습니다.

다소 느린 시뮬레이션 속도를 개선할 필요가 있습니다.

seong-hun, 2019-09-10:
`main_test_PID.py` 테스트 결과 `QuadrotorHoveringEnv.get_ob` 메소드의 `scipy.spatial.transform.Rotation` 이 상당한 속도 저하를 야기하는 것 같습니다.

`Rotation`을 사용할 때: 약 8초
사용하지 않을때: 약 4초

seong-hun, 2019-09-12:
`scipy.integrate` 모듈을 사용하면 더욱 빠르게 할 수 있지 않을까요?

seong-hun, 2019-10-17:
`scipy.integrate` 모듈의 `solve_ivp`, `odeint`, 그리고 `fym.core` 의 `rk4` 속도 비교

- 스크립트:

```python
from scipy.integrate import odeint, solve_ivp
from scipy.special import gamma, airy
from fym.core import rk4

y1_0 = 1.0 / 3**(2.0/3.0) / gamma(2.0/3.0)
y0_0 = -1.0 / 3**(1.0/3.0) / gamma(1.0/3.0)
y0 = [y0_0, y1_0]
t = [0, 0.005, 0.01]

def func(t, y):
    return [t*y[1], y[0]]
```

- 결과:

```python
In [1]: %timeit rk4(lambda y, t: func(t, y), y0, t)
63.6 µs ± 1.01 µs per loop (mean ± std. dev. of 7 runs, 10000 loops each)

In [2]: %timeit odeint(lambda y, t: func(t, y), y0, t)
39.6 µs ± 299 ns per loop (mean ± std. dev. of 7 runs, 10000 loops each)

In [3]: %timeit solve_ivp(func, [0, 0.01], y0, t_eval=t)
254 µs ± 1.39 µs per loop (mean ± std. dev. of 7 runs, 1000 loops each)
````

## 결론

`scipy.integrate.odeint` 가 가장 빠르다.

seong-hun, 2019-10-19:
#100 에서 해당 이슈를 해결하였으므로 닫겠습니다.

closed 2019-10-19 by seong-hun