seong-hun/fym #85: `develop` 브랜치와 `master` 브랜치의 통합 (git flow 에서 github flow 로)

seong-hun, 2019-09-04:
현재 `develop` 브랜치를 따로 두고 release 를 `master` 브랜치에서 시행하고 있는데, 우리 작업 상 조금 번거롭지 않나 싶습니다. 항상 일을 할 때를 보면 `develop` 브랜치에서 하게 되구요. 

그래서 이런 단점을 보완한 [Gtihub flow](https://dogfeet.github.io/articles/2011/github-flow.html) 를 적용해보면 어떨까 싶습니다. 지금이랑 거의 비슷한데 그냥 `develop` 브랜치와 `master` 브랜치가 합쳐진 것이라 보면 되겠습니다.

## 차이점
- 브랜치 이름을 명확히 적음 (`fix-HandReach-v0-pickling`)
- 브랜치를 만들고 커밋을 여러개 할 수 있음
  - 대신 풀리퀘스트를 보내고 `master` 브랜치에 머지를 할 때, `Squash and merge` 옵션 선택
  - 이렇게 하면, `master` 브랜치에는 머지된 브랜치의 상세 커밋들이 보이지 않고 하나의 새로운 커밋만 남음
  - 그리고 머지된 브랜치를 삭제 (`git log` 에도 보이지 않으나 커밋은 존재, SHA로 검색가능)
  - 풀리퀘스트 페이지 (Github 상) 에서는 상세 커밋들을 볼 수 있음
- 리베이스를 할 필요 없음

**Github flow 적용 레포들**
- [Facebook React](https://github.com/facebook/react)
- [OpenAI Gym](https://github.com/openai/gym)

JinraeKim, 2019-09-04:
기존 방식이 다소 번거로워서 찬성합니다

seong-hun, 2019-09-09:
v0.2.2 부터 이 방식으로 진행합니다.

closed 2019-10-19 by seong-hun