seong-hun/fym #130: [Parallel computing] pickle issue

JinraeKim, 2020-01-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
강화학습 데이터 샘플링 효율을 높이기 위해 병렬연산을 수행하려는중, 현재의 `core.py` 구조에서는 어려움을 확인하였습니다.
-  pickle 시 dict_values objects 는 지원하지 않음
    - 이는 `core.py` 의 class `BaseEnv` 에서, self.systems 를 dict_values 로 받지 않으면 개선 가능할 것으로 보입니다 (실제로 실행시 발생하는 다음 에러를 해소해줌:
[image]
).
- pickle 전후로 데이터 형태가 같아야 함 (?)
    - `core.py` 에서 `        self.systems = list(systems_dict.values())
` 를 통해 위의 `dict_values` 에러를 해소하여도 다음 에러가 발생합니다:
[image]

이 에러는 다음 [링크](https://stackoverflow.com/questions/1412787/picklingerror-cant-pickle-class-decimal-decimal-its-not-the-same-object) 에서 언급된, pickle 전후의 데이터 형태 유지에 관한 내용으로 예상됩니다.

관련 코드 예시는 forked repo 의 branch [issue-parallel-computing](https://github.com/JinraeKim/fym/tree/issue-parallel-computing) 에서 `test/parallel.py` 로 올려두었으니 참고해주세요..

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
강화학습 데이터 샘플링 효율을 높이기 위해서는 병렬연산을 고려해서 `fym` 을 확장하면 좋겠습니다.
찾아본 바로는 module `multiprocessing` 을 많이 쓰는 것 같은데, 이것과 잘 호환될 수 있게 확장하는 것을 희망합니다.
솔직하게 `Python` 에서 병렬연산을 수행해본 경험이 많지 않아, 좋은 해결방안은 잘 떠오르지가 않습니다..

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

JinraeKim, 2020-01-30:
계속 붙잡다가 해결이 안되어 도움 요청..!

JinraeKim, 2020-01-30:
## Note
Module `ray` 가 기존의 `multiprocessing` 의 대체제로 사용될 수 있다고 하여 찾아보았는데요 (뭐 좀 더 빠르고 좋다네요..?).
해당 모듈을 사용하였을 때
1) `dict_values` 를 지원하지 못하는 이슈는 동일하게 발생하였으나
2) pickle 전후로 데이터 형태가 같아야 하는 오류는 발생하지 않았습니다.
또한, 꽤 직관적이라 parallel computing 하기가 용이해보입니다.
`ray` 를 이용하여 돌아가는 코드를 동일 브랜치의 "Change parallel loop using ray" 라는 이름의 커밋으로 달아두었으니 참고 바랍니다.

여기에 [ray 관련 레포](https://github.com/ray-project/ray) 를 달아둡니다.
- Ray 설명
    - Ray is packaged with the following libraries for accelerating machine learning workloads.

seong-hun, 2020-02-03:
multiprocessing pickle 문제는 함수를 전역으로 만들어 둬야 한다고 합니다. [참고](https://stackoverflow.com/a/52283968)

함수를 전역으로 만들고 다시 해보세요.

seong-hun, 2021-05-12:
해결 된건가요?

JinraeKim, 2021-05-12:
> 해결 된건가요?

잘 기억나지 않네요 ㅠ 이미 저 당시랑 현재랑 fym이 바뀌어서 닫으셔도 될듯

closed 2021-05-12 by seong-hun