seong-hun/fym #165: logger 모듈 버그

JungYT, 2021-05-04:
[image]

현상
1. eval_logger에 state만 저장할 때는 문제 없이 작동 (eval_logger.record(**info))
2. DDPG의 actor와 critic의 DNN parameters 저장하는 코드를 추가
eval_logger.record(actor=agent.target_actor.parameters())
eval_logger.record(critic=agent.target_critic.parameters())
3. 4번 째 에피소드 진행할 때 에러 발생. 사진을 보면 에러가 발생한 위치는 eval_logger.record(**info)로 state를 저장하는 곳

추측
사진을 보면 logger의 flush 함수에서 문제 발생
아마도 parameter를 저장하면서 한 번에 저장하는 데이터의 길이가 max_len을 넘어가면서 생기는 문제?

seong-hun, 2021-05-04:
제 생각에는 데이터 타입이 현재 `fym.logging` 모듈에서 지원하지 않는것 같습니다. 
예를 들어 `None` 값들은 저장이 안돼요.. (HDF5 파일 특성)

혹시 DNN parameters의 데이터 타입이 뭔지 알 수 있을까요?

JungYT, 2021-05-04:
DNN parameter를 저장하는 logger를 별도로 만들어 저장한 결과 에러가 발생하지 않습니다. 데이터 타입의 문제는 아닌 것 같습니다.

JungYT, 2021-05-04:
오잉 저장한거 불러오니까 {} 네? 다시 해보겠습니다.

seong-hun, 2021-05-04:
저 에러를 디버깅 모드로 들어가서 아래 부분에서 `val`이 뭔지 확인해주세요
[image]

JungYT, 2021-05-04:
[image]
같은 코드를 돌렸는데 에러 재현이 안 됩니다..
대신 agent.target_actor.parameters() 가 이상하다는 건 확인했습니다. 이 부분이 문제인 것 같습니다.

seong-hun, 2021-05-04:
저녀석은 `list()`를 씌워주면 될것 같군욥.. 혹은 PyTorch에 파라미터 값들을 딕셔너리로 내뱉는게 있었던거 같은데..

JungYT, 2021-05-04:
pytorch 자체에서 지원하는 save를 사용하려고 합니다.
참고로, list() 씌우면 에러나네요

seong-hun, 2021-05-04:
그게 맘 편할듯 ㅋㅋ `state_dict` 메소드도 에러나나요?

seong-hun, 2021-05-04:
그냥 [PyTorch 기능](https://tutorials.pytorch.kr/recipes/recipes/saving_and_loading_a_general_checkpoint.html) 쓰는게 편할것 같습니다.

JungYT, 2021-05-04:
https://tutorials.pytorch.kr/beginner/saving_loading_models.html
https://justkode.kr/deep-learning/pytorch-save
이거 씁시다ㅋㅋ 이슈 닫겠습니다.

closed 2021-05-04 by JungYT