seong-hun/fym #166: Documentation; methods in `fym.utils.rot`

JinraeKim, 2021-05-12:
In `fym.utils.rot`, some methods are provided, which are related to rigid body attitude representation.

However, there is no description about "from which to which" coordinate systems, such as inertial coord. to body coord.
I request docstrings for them.

Note: It seems that all unit quaternion, DCM, and Euler angles represent body attitude w.r.t. inertial frame. Also, Euler angles are related to ZYX angles, i.e., yaw, pitch, roll.

seong-hun, 2021-05-12:
It seems that you need a simple description for what coordinates the DCM transforms.
However, there already exists [a comment](https://github.com/fdcl-nrf/fym/blob/c40448974dd4480ea4135aebb707542145f85c70/fym/utils/rot.py#L7) on that.
Isn't is sufficient?

JinraeKim, 2021-05-12:
> It seems that you need a simple description for what coordinates the DCM transforms.
> However, there already exists [a comment](https://github.com/fdcl-nrf/fym/blob/c40448974dd4480ea4135aebb707542145f85c70/fym/utils/rot.py#L7) on that.
> Isn't is sufficient?

It's not sufficient.
1. One may be confused with if """DCM: C_{b/i}"""  represents "from b to i" or "i to b". It is merely a conventional representation.
2. No description of the order of Euler angles, e.g., ZYX (first Z applied), XYX, XYZ, etc.
3. No description of the unit quaternion.

seong-hun, 2021-05-12:
> * No description of the order of Euler angles, e.g., ZYX (first Z applied), XYX, XYZ, etc.

[It exists.](https://github.com/fdcl-nrf/fym/blob/c40448974dd4480ea4135aebb707542145f85c70/fym/utils/rot.py#L63)

> 1. It is merely a conventional representation.

It's conventional... at least in aerospace literatures. However, for users who are not familiar to aerospace control, I'll add some references for them.

> 3\. No description of the unit quaternion.

There is no need to explain what unit quaternion is for rotational representation.

JinraeKim, 2021-05-12:
Well, if you think so, I don't wanna rush you.

seong-hun, 2021-05-12:
If you are confused, please clarify how the explanation can be misunderstood and please suggest any additional description that satisfy you through a PR.

JinraeKim, 2021-05-12:
> If you are confused, please clarify how the explanation can be misunderstood and please suggest any additional description that satisfy you through a PR.

My pleasure :)

seong-hun, 2021-05-12:
#168
Now I understand why you said current docstring is insufficient... Thanks a lot.

PR #168 (2021-06-07): Add docs for `fym/utils/rot.py`
Resolve #166.

자세 표현하는게 자주 헷갈려서 변수에 대한 설명 및 동역학을 docstring에 적어보았습니다.

@seong-hun 제가 docstring 을 자주 안 써봐서, 혹시 convention 과 다르다면 리뷰 부탁드립니다 ㅎ
closed 2021-06-07 by seong-hun