seong-hun/fym #179: fym update 후 다음과 같은 에러가 생겼습니다.

JungYT, 2021-06-29:
[image]

logging 단계에서 문제가 생긴거로 보입니다. 10개 주기로 데이터에 발산이 생깁니다.

코드 링크는 아래와 같으며, main.py를 돌리면 되고 dynamics.py를 보시면 됩니다. 그래프는 utils.py를 보면 됩니다.

https://github.com/JungYT/multi-transportation

seong-hun, 2021-06-29:
발산이 아니라 empty 초기화 된 에레이에 값이 제대로 안 들어간 것처럼 보이네요.. dot 값은 모든 시스템에 대해 잘 주고 있나요?

JungYT, 2021-06-29:
좀 더 확인해보니 5초간 시뮬레이션, ode_int_step 이 10이므로 데이터 개수가 500이어야 하는데 550개가 있습니다. 10개 기준으로 한 번씩 이상한 값이 추가로 logging되고 있네요.

JungYT, 2021-06-29:
[image]

time 저장에도 다음과 같이 10개 간격(아마도 ode_int_step과 동일한 간격)으로 이상한 값이 들어가있습니다.
set_dot에서 모든 시스템에 대해 set_dot을 call하고 있습니다.

seong-hun, 2021-06-29:
음 보니까 문제가 된다면 이 부분일 수도 있을 것 같습니다.
env에 logger를 등록하면 내부로거로 동작하는데, step 외부에서 또 record를 시도하는군요..
https://github.com/JungYT/multi-transportation/blob/3431b66ec353c459c770c8fc53a49f7496b90113/main.py#L294

JungYT, 2021-06-29:
오잉 외부로거는 아예 따로 만들었던 것 같은데...왜 저렇게 되어있지.. 수정해보겠습니다.

JungYT, 2021-06-29:
해당 문제가 맞네요. 버그 리포트는 클로즈 하고 새 이슈 파겠습니다ㅎㅎ

closed 2021-06-29 by JungYT

seong-hun, 2021-06-29:
감사합니다~

PR #180 (2021-06-29): Block direct call of `Logger.record` method of inner Logger
Inner ``Logger``s was designed to record inner data of ``BaseEnv``, so attempts to directly record any data to the inner logger should be blocked.

I've made a method ``Logger._record`` that works the same as the previous ``Logger.record`` method, and add a line to check the ``Logger`` is inner or not inside the previous ``Logger.record`` method for compatibility.

This resolves #179 .