seong-hun/fym #181: logging multi rate data

JungYT, 2021-06-29:
variable step(예: ode_int_step=10)로 시뮬레이션 할 때 multi rate 값들을 logging하고 싶습니다.

예를 들어서,

0.1(dt)마다 action을 넣고, reward를 뽑지만 적분은 variable step으로 해서 0.01초마다 state를 저장해 그래프를 그리고 싶은 상황.

state는 0.01초 간격으로 저장해야 되지만(logger_callback 혹은 set_dot에서 return을 통해서 가능) action과 reward는 0.1초 간격으로  저장하고 싶습니다.

action의 경우 0.1초 logger_callback에서 (0.1초 동안) 동일한 값을 그냥 0.01초 간격으로 저장해도 괜찮지만,
reward는 0.1초 간격의 state를 이용해 계산되야 하지만 logger_callback에서 reward를 계산해서 저장할 경우 0.01초 간격의 state를 이용해 계산된 값을 저장하게 됩니다.

되면 좋고 안 되면 어쩔 수 없고..

seong-hun, 2021-06-29:
음 이거는 의도적으로 이렇게 구현한 것인데.. 한 파일에 저장이 되면 편한점도 있겠지만 불편한 점도 있을듯 합니다.
예를 들어서, 내부 state와 외부 state (reward 등)들을 한 파일에 저장하면, 그 파일을 load 했을 때 각각의 key가 time step을 따라가는지, iteration step을 따라가는지 데이터를 생성했던 코드를 보지 않는 이상 구분할 수 없습니다.

하지만 여전히 한 파일에 저장하고 싶다고 한다면 직접 두 dictionary 들을 합쳐서 저장할 수 있습니다. 중간중간 할 수도 있겠고 후처리를 할 수도 있겠지만 저는 깔끔한 후처리를 선호합니다. 다음은 예시 코드입니다.

```python
env.logger = fym.Logger("env-data.h5")
logger = fym.Logger("agent-data.h5")

obs = env.reset()
for i in range(1000):
    action  = agent.get(obs)
    next_obs, done = self.step(action):
    logger.record(i=i, obs=obs, action=action, next_obs=next_obs)
    if done:
        break
    obs = next_obs

env.close()
logger.close()

fym.save("new-data.h5", dict(fym.load(env.logger.path), **fym.load(logger.path)))
# In Python 9: fym.save("new-data.h5", fym.load(env.logger.path) | fym.load(logger.path))
```

JungYT, 2021-06-29:
오호 그렇군요. 타당한 것 같습니다. 어느정도 해결책도 알았으니 이슈 닫겠습니다.

closed 2021-06-29 by JungYT